equinor/flotilla #12: Create Start Mission Button

vetlek, 2021-12-03:
The Mission Overview of available missions in [figma](https://www.figma.com/file/q6vsNRogzV6r9vTCCDsocd/First-Draft?node-id=0%3A1) have a play button that is used to start the connected mission. Clicking the button should open an popover menu where the user can select which of the available robots to use for the mission and then press play to activate the mission. A basic implementation of this button should be created. The button don't need to be able to start the mission, but should display the described information and interfaces. Connected to #7.

aestene, 2022-03-05:
This issue has automatically been marked as stale as there has been no activity for 60 days. Remove stale label or comment or this will be closed in 7 days.

nicholasdalhaug, 2022-03-07:
We will reevaluate once we have a clearer UX plan.

closed 2022-03-07 by nicholasdalhaug