equinor/flotilla #7: Create Mission Information Component

vetlek, 2021-12-03:
The Available Mission Overview Component in [figma](https://www.figma.com/file/q6vsNRogzV6r9vTCCDsocd/First-Draft?node-id=0%3A1) displays a list of all the available missions on a certain asset. Each item in the list corresponds to a single mission and displays some basic information about the mission together with a start mission button.

Create a Mission Information Component that displays all the information as described in [figma](https://www.figma.com/file/q6vsNRogzV6r9vTCCDsocd/First-Draft?node-id=0%3A1). The play button should not work.

aestene, 2022-03-07:
This issue has automatically been marked as stale as there has been no activity for 60 days. Remove stale label or comment or this will be closed in 7 days.

nicholasdalhaug, 2022-03-07:
We will reevaluate once we have a clearer UX plan.

closed 2022-03-07 by nicholasdalhaug