equinor/flotilla #26: Remove '$' from README

anetteu, 2022-01-12:
Remove '$' from README.md such that when copying commands, you can paste them into a terminal and execute directly.

pramod74, 2022-01-13:
Hi @anetteu,

I would like to work on this issue. Can you please assign the issue to me and guide me on what needs to be done? I see three README.md files in the repository. Is this change required in all of them?

Thanks,
Pramod

anetteu, 2022-01-13:
Hi @pramod74, 

Thank you for reaching out, we appreciate that you would like to contribute to our project. It looks like this issue was solved yesterday, and I will therefore close this issue.

Please consider assigning to another issue to contribute. We would be happy to guide you. 

If you do not find anything interesting in this repository at the moment, I encourage you to come back later. You could also checkout other open source repositories from Equinor or other organizations. As an example, you could have a look at [isar](https://github.com/equinor/isar/issues), which is developed by the same team as flotilla. 

Good luck!

closed 2022-01-13 by anetteu

pramod74, 2022-01-13:
Sure, @anetteu. Thanks!