equinor/flotilla #48: Establish timeseries database for Flotilla

aestene, 2022-01-25:
**Describe the new feature you would like to see**
The Flotilla system will have access to timeseries streamed from the MQTT broker which should be stored in a timeseries database. Examples are mission status, battery level, pose etc.

**Describe the solution you'd like**
The MQTT broker established as part of #47 should stream all relevant topics to the database where it is stored in a functional format. An example of a timeseries database is [InfluxDB](https://www.influxdata.com/products/influxdb-cloud/?utm_source=google&utm_medium=cpc&utm_campaign=2020-09-03_Cloud_Traffic_Brand-InfluxDB_INTL&utm_term=influxdb&gclid=CjwKCAiA3L6PBhBvEiwAINlJ9N5oGarlSv-bCm3-R3EPa3qjQtixscADBzAj7cO3ENeEnudKVPuHRhoC-4EQAvD_BwE) which is available in Azure.

**How will this feature affect the current Threat Model?**
Access to the timeseries database must be managed.
- Note that the first iteration will only be a database available locally on the facility tightly coupled with Flotilla.

aestene, 2022-01-25:
[image]

aestene, 2022-05-20:
Alternatives mentioned have also been IOTHUB, MQTT + SparkPlug

GodVenn, 2023-04-28:
While looking into this, I think that PostreSQL and timescale db is a good option for us.
By using that we can link the timeseries relationally to the rest of our database.

https://azure.microsoft.com/nb-no/blog/power-iot-and-time-series-workloads-with-timescaledb-for-azure-database-for-postgresql/
https://docs.timescale.com/getting-started/latest/

PR #805 (2023-10-03): Add timeseries database
Closes #48 

Changes the database type to PostgreSql
Adds timeseries to the database
Updates them based on MQTT

Loosely followed this guide for the setup
https://gibinfrancis.medium.com/timescale-db-with-ef-core-94c948829608
while also using documentation from here:
https://github.com/timescale/timescaledb
closed 2023-10-03 by Christdej