equinor/flotilla #50: Handle refresh tokens for authentication

aestene, 2022-01-27:
**Describe the bug**
If the user authenticates in the OpenAPI documentation the authentication token will not be automatically refreshed after 1 hours (token timeout). This should be handled. 

**To Reproduce**
Steps to reproduce the behavior:
- Authenticate in OpenAPI
- Wait 1 hour
- MAke a request

**Expected behavior**
Instead of failing authentication a refresh token is used to authenticate again and from the users POV there is no change.

**Screenshots**
[image]

vetlek, 2022-02-24:
Seems like there is no support for refresh tokens in the library that is currently used for authentication.

vetlek, 2022-02-25:
Is this a real issue though? The API is not supposed to be used from the SwaggerUI except for simple testing in the development, and there is no issues with revoked access tokens when the API is used by clients with secrets. Can't really see the problem with having to authenticate yourself every 60-90 minutes.

closed 2022-03-11 by aestene