equinor/flotilla #87: Document MQTT endpoints through Async API

aestene, 2022-02-14:
**Describe the new feature you would like to see**
AsyncAPI is a tool similar to OpenAPI which may be used to document subscribe and publish options for a message broker like MQTT. This could be used to document which topics Flotilla expects to receive publishes too and which topics are available to be subscribed on. https://www.asyncapi.com/

**Describe the solution you'd like**
The AsyncAPI documentation is the go to location for figuring out which topics should have relevant information on them.

**How will this feature affect the current Threat Model?**
N/A

vetlek, 2022-02-14:
Move this issue to flotilla-openapi repo?

vetlek, 2022-02-17:
[PR in Flotilla-Openapi](https://github.com/equinor/flotilla-openapi/pull/8)

aestene, 2022-02-21:
PR has been closed

closed 2022-02-21 by aestene