equinor/flotilla #93: Setup Access Control List for the MQTT-broker

vetlek, 2022-02-24:
**Describe the improvement you would like to see**
The security of the broker can be improved by using authorization on the different topics. The mosquitto broker offers the opportunity to use an Access Control List (ACL) to perform authorization. In ACL you can specify which topics a certain user can publish/subscribe to. A brief introduction can be found [here](https://www.chirpstack.io/project/guides/mqtt-authentication/). Different users (with passwords) should be generated for the different roles (streaming of telemetry, status from isar, flotilla backend, flotilla frontend).

**How will this change existing functionality?**
This will improve the security of the broker and will have potential effects to all clients of the broker, both publishers and subscribers.

**How will this improvement affect the current Threat Model?**
This should reduce the risk of unauthorized publishing to the broker and leak of information (listening to certain topics).

aestene, 2022-04-26:
This issue has automatically been marked as stale as there has been no activity for 60 days. Remove stale label or comment or this will be closed in 7 days.

tsundvoll, 2022-05-18:
Related to #127

PR #228 (2022-06-08): Authentication for mqtt
Closes #93 
closes #127