equinor/flotilla #138: Reuse specification of responses on endpoints

aestene, 2022-04-20:
**Describe the improvement you would like to see**
Currently responses are set on every endpoint by giving status code, model and a description. These are however the same all over the place where you want a 404 - Not Found for example. It should be possible to have a system where standard responses can be reused across endpoints so these are not duplicated everywhere.

**How will this change existing functionality?**
There should be no effect on current functionality but the code should be more organized.

**How will this improvement affect the current Threat Model?**
No effect.

aestene, 2022-04-20:
[image]
This pattern is present on every endpoint.

anetteu, 2022-04-21:
I think the attached screenshot is outdated? The status code and description are the same several places, but the model is different depending on the endpoint. Are the responses similar enough to make a generalized response that can be used across endpoints?

aestene, 2022-04-25:
They are similar enough for the `UNAUTHORIZED`, `NOT_FOUND` and `CONFLICT` codes. The success code may vary but the others are always the same.

anetteu, 2022-04-25:
Ok, I see that this issue is now more relevant with the commits merged after my previous comment

closed 2022-05-02 by aestene