equinor/flotilla #195: Add framework for integration testing

aestene, 2022-05-10:
**Describe the new feature you would like to see**
In #174 it was attempted to write tests for the `missions` controller, however it was found that implementing an integration test framework where the application as a whole is spun up could be a better solution than manually mocking the services.

**Describe the solution you'd like**
Follow Microsofts recommended setup: https://docs.microsoft.com/en-us/aspnet/core/test/integration-tests?view=aspnetcore-6.0

**How will this feature affect the current Threat Model?**
Should be positive as it enables us to test the application as a whole.

aestene, 2022-10-03:
This issue has automatically been marked as stale as there has been no activity for 60 days. Remove stale label or comment or this will be closed in 7 days.

oysand, 2022-11-07:
Done

closed 2022-11-07 by oysand