equinor/flotilla #197: Enum's returned through the swagger should display the value and not the int.

oysand, 2022-05-11:
**Describe the bug**
RobotStatus and EventStatus now display the int representation from the enum rather than the value.

**To Reproduce**
Call the /events or /robots endpoint through the swagger inteface. 

**Expected behavior**
The status should be displaying the value. For instance the EventStatus should show Pending rather that 0.

**Screenshots**
If applicable, add screenshots to help explain your problem.

anetteu, 2022-05-11:
With the current implementation, the enum variables are represented by ints in the database and therefore represented as ints to the user through json/swagger. This is not really a bug, but it is a good idea to change the implementation to provide string values of the enums also in the database for better readability. 

There is a discussion about implementing this differently so the enum values is represented as strings also in the database and in the json provided to the user in currently open PRs. The proposed solutions are adding a json serializer to the controller (#183) or a different implementation of the enums (#194 ). This issue is expected to be solved by those PRs.

anetteu, 2022-05-12:
Solved by #183

closed 2022-05-12 by anetteu