equinor/flotilla #211: Look into importing MQTT message models from ISAR to Flotilla

GodVenn, 2022-05-19:
**Describe the new feature you would like to see**
Currently we are "copying" the MQTT message models from ISAR into classes in flotilla backend.
A possible improvement to this could be to somehow import them.

**Describe the solution you'd like**
We could maybe include the models in versioning, giving alerts if the model has changed in ISAR when attempting to communicate from Flotilla with outdated models.

**How will this feature affect the current Threat Model?**

vetlek, 2022-05-20:
A possible solution is using something like [Protobuf](https://developers.google.com/protocol-buffers) to define messages and generate models. This is used in MQTT Sparkplug to define messages (Sparkplug might be an opportunity in itself, but have some overhead that might not be necessary).

GodVenn, 2022-08-08:
Not needed

closed 2022-08-08 by GodVenn