equinor/flotilla #226: Flotilla misses the first updates from ISAR

aestene, 2022-06-07:
**Describe the bug**
Flotilla will as of #224 miss the initial updates to mission, task and step status because ISAR starts the mission and fires the first MQTT messages before Flotilla is finished writing the initial report to the database. This is because the mission starts before ISAR returns the mission details to Flotilla.

**To Reproduce**
Start a mission in ISAR from Flotilla. Easiest seen using isar-robot as all updates will fail because it's so fast.

**Expected behavior**
Flotilla should receive and react to all updates once a mission has started.

aestene, 2022-06-21:
Sleep has been put in place as a temporary solution to this issue. https://github.com/equinor/isar/commit/ce189ada782529ce75e2b5cd67ca6b58ee4b284e

aestene, 2022-08-21:
This issue has automatically been marked as stale as there has been no activity for 60 days. Remove stale label or comment or this will be closed in 7 days.

MortFred, 2022-08-22:
Will be resolved by #274

Christdej, 2023-02-13:
Remove workaround https://github.com/equinor/isar/commit/ce189ada782529ce75e2b5cd67ca6b58ee4b284e

aestene, 2023-02-27:
Did you try to remove this workaround @Christdej? As far as I can tell it is still there. Until further notice I will move this issue back to blocked.

Christdej, 2023-02-28:
Did not test, just saw that in the comments Morten wrote that it would be resolved by #274 , but apparently it was not.

GodVenn, 2023-04-14:
We believe this can be removed now

aestene, 2023-06-22:
Resolved by https://github.com/equinor/isar/pull/477

closed 2023-06-22 by aestene