equinor/flotilla #280: Handle case when scheduled mission fails to start

GodVenn, 2022-07-14:
**Describe the new feature you would like to see**
When a scheduled mission currently fails to start, it is retried instantly and infinitely. 
We should figure out how to handle scheduled missions failing. 

**Describe the solution you'd like**
Some options:
- Retrying a couple of times, and then marking them as failed and leaving it to the end-user to decide (frontend)
- Moving them to a later time slot
- Panic

**How will this feature affect the current Threat Model?**

sondreo, 2022-08-01:
Mark as failed, and leave to user.

anetteu, 2022-08-15:
Blocked,  awaiting database context fix #322

MortFred, 2022-08-22:
Fixed by #322, should be tested.

GodVenn, 2022-10-03:
This has been solved by a combination of various pull requests.
Failed missions are now marked as failed in the database without re-attempting to start them

closed 2022-10-03 by GodVenn