equinor/flotilla #324: ScheduledMissions and robots only available after authorizing backend

oysand, 2022-08-17:
**Describe the bug**
When starting flotilla the scheduled missions did show. However after going to https://localhost:8000/swagger/index.html and clicking authorize there the scheduled missions started and showed.

**To Reproduce**
Start the broker in docker, start the backend outside docker and start the frontend outside docker. Open http://localhost:3001/ and see that the robots and shceduled missions does not show. Open https://localhost:8000/swagger/index.html, click authorize and check http://localhost:3001/ again.

**Expected behavior**
It should not be necessary to authorize the backend in addition to login into Flotilla.

**Screenshots**
If applicable, add screenshots to help explain your problem.

GodVenn, 2022-08-18:
Is there a call to the backend in the frontend console log with a 401 status code returned?

GodVenn, 2022-08-23:
I did the steps to reproduce and could not reproduce the issue.
@oysand could you verify that this is still an issue?

oysand, 2022-09-06:
I've had mixed experiences, it seems that sometimes this happens, but not consistently. Usually it happens when it has been some time between starting the application

aestene, 2022-09-19:
@eivindsjovold will have a look at recreating this bug

eivindsjovold, 2022-09-29:
Snippet from browser console when this bug occurs: 

[image]

eivindsjovold, 2022-10-27:
I agree @GodVenn.

closed 2022-10-31 by oysand