equinor/flotilla #357: Time is wrong in Frontend

MortFred, 2022-09-21:
**Describe the bug**
Time for mission start and end is wrong in frontend

**To Reproduce**
Load the frontend and compare with API times

**Expected behavior**
Time in frontend should represent user's local time.

**Screenshots**
Frontend:
[image]

From Backend (in UTC):
[image]

aestene, 2023-01-10:
This issue has automatically been marked as stale as there has been no activity for 60 days. Remove stale label or comment or this will be closed in 7 days.

Eddasol, 2023-04-22:
Already fixed

closed 2023-04-22 by Eddasol