equinor/flotilla #378: Hold the latest robot status in a variable in backend to display to frontend

tsundvoll, 2022-10-03:
**Describe the new feature you would like to see**
This is to avoid to have the timeseries database. It might be a temporary solution (or possibly the way to do it later)

**Describe the solution you'd like**
The database should still be updated, but the frontend can query from the variable instead of the database.

At startup, fetch the latest status from the database.

**How will this feature affect the current Threat Model?**

GodVenn, 2022-10-03:
I guess what we are essentially talking about is "caching" the robot status in the backend. 
Might be worth looking into if ASP.NET or Entity Framework somehow support this natively.

aestene, 2022-12-03:
This issue has automatically been marked as stale as there has been no activity for 60 days. Remove stale label or comment or this will be closed in 7 days.

Christdej, 2023-05-08:
Not needed

closed 2023-05-08 by Christdej