equinor/flotilla #382: Add config to reduce (set) log-level in console

tsundvoll, 2022-10-06:
**Describe the improvement you would like to see**
For all three components (frontend, broker and backend), find a way to reduce log-level to console.

**How will this change existing functionality?**
Today, there is an overwhelming amount of logging to console, this change would help sort out what is important.

**How will this improvement affect the current Threat Model?**

GodVenn, 2022-10-20:
This can be done in backend by changing the Appsettings file for the relevant environment

aestene, 2022-12-20:
This issue has automatically been marked as stale as there has been no activity for 60 days. Remove stale label or comment or this will be closed in 7 days.

Christdej, 2023-05-08:
Solved

closed 2023-05-08 by Christdej