equinor/flotilla #389: Remove dummy missions

oysand, 2022-10-12:
**Describe the improvement you would like to see**
When starting the backend the database is populated with dummy missions. It should be blank to begin with. These missions are integral to the tests and should be mocked for the tests.

**How will this change existing functionality?**
Missions used in the tests should only be present when testing.

**How will this improvement affect the current Threat Model?**
N/A

oysand, 2022-10-12:
#390 temporarily removes the affected tests and comments out the adding of the missions in initDb. This is only a temporary fix and should be addressed properly

aestene, 2022-12-11:
This issue has automatically been marked as stale as there has been no activity for 60 days. Remove stale label or comment or this will be closed in 7 days.

oysand, 2023-01-09:
Missions improved sufficiently

closed 2023-01-09 by oysand