equinor/flotilla #394: In the mission view, sort the tasks in correct order

tsundvoll, 2022-10-12:
**Describe the bug**
The tasks on a mission is not displayed in the order they are executed

**To Reproduce**
Schedule a mission with more than one task. When the mission is started, the displayed order is not necessarily the order the tasks will be run.

**Expected behavior**
The tasks are displayed in the order they are executed.

**Screenshots**
If applicable, add screenshots to help explain your problem.

aestene, 2022-12-12:
This issue has automatically been marked as stale as there has been no activity for 60 days. Remove stale label or comment or this will be closed in 7 days.

Eddasol, 2023-04-22:
Sufficiently covered by issue #766

closed 2023-04-22 by Eddasol

GodVenn, 2023-04-25:
Actually closed by #607