equinor/flotilla #396: Document / clean up the different ID terms used

tsundvoll, 2022-10-13:
**Describe the new feature you would like to see**
Background: The backend operates with many different ID's. MissionID, TaskID, StepID. Some connected to Echo, some to ISAR and some in addition.

Also, when retreiving missions from the backend's /missions, `"isarMissionId": null,` indicates that the isarMissionId is not returned. Consider if this is something we need.

**Describe the solution you'd like**
This is a suggestion to have a look on those different ID's and see if something can be made clearer.

**How will this feature affect the current Threat Model?**

aestene, 2022-12-12:
This issue has automatically been marked as stale as there has been no activity for 60 days. Remove stale label or comment or this will be closed in 7 days.

tsundvoll, 2023-01-11:
This has been improved on, closing the issue

closed 2023-01-11 by tsundvoll