equinor/flotilla #402: Filter away offline robots in robot status list

aestene, 2022-10-14:
**Describe the improvement you would like to see**
Currently all robots in the database are displayed in the "Robot status" list. However, the offline robots should not be displayed. A possible solution is to add an optional parameter to the /robots endpoint in the backend where one may specify status. Otherwise the easiest and maybe good enough solution is to just filter in the frontend saying that robots with status `Offline` should not be displayed.

[image]

**How will this change existing functionality?**
Only the relevant robots are displayed in the frontend.

**How will this improvement affect the current Threat Model?**
Not affected.

aestene, 2022-12-13:
This issue has automatically been marked as stale as there has been no activity for 60 days. Remove stale label or comment or this will be closed in 7 days.

oysand, 2023-01-09:
Display Offline, but introduce Decommissioned state

aestene, 2023-01-09:
Related to https://github.com/equinor/isar/issues/341
This will enable us to set offline in a new way. In addition we can implement decommisioned as a new state which is set manually and thus filtered away.

Christdej, 2023-02-13:
Remove "Enabled" field on Robot model

Christdej, 2023-05-08:
Solved

closed 2023-05-08 by Christdej