equinor/flotilla #404: Robot is displayed as `Offline` instead of `Busy`

aestene, 2022-10-14:
**Describe the bug**
When a mission is scheduled on a robot and it is `Busy` the robot is displayed as `Offline` rather than `Busy`.

**To Reproduce**
Start Flotilla, isar-turtlebot and ISAR and schedule a mission for R2-D2 from the frontend. The robot should show as offline. 

**Expected behavior**
Expect an indication that the robot is busy or in a mission. For example the status `Busy` showing up in a red color. 

**Screenshots**
 In this picture a mission is running on R2-D2 but it is displayed as offline.
[image]

martinko2369, 2022-10-14:
Hi @aestene, i would like to work on this issue.

oysand, 2022-10-31:
Done

closed 2022-10-31 by oysand