equinor/flotilla #408: Scheduling multiple missions with remote database means only one mission is executed

aestene, 2022-10-14:
**Describe the bug**
Scheduling a lot of missions simultaneously using the remote database the robot is not updated to available after a mission has been executed. Thus leaving the rest of the missions as pending. 

**To Reproduce**
1. Start the full deployment stack with the backend in the Staging environment. Bug was seen with isar-turtlebot and has been seen using an actual robot.
2. Schedule multiple missions on the same robot.
3. Wait for the first mission to finish and see the effect.

**Expected behavior**
Once a mission is complete the robot should be set as available and queue the next mission. Eventually all missions should be completed.

**Screenshots**
Log from Flotilla backend showing that the message from ISAR that the mission is complete har been received and that the robot should have been set as available.
[image]

Robot `william` is set as "Offline"
[image]

Remaining missions on `william` are not started as the robot is marked as busy. 
[image]

aestene, 2022-10-14:
Maybe have a look @oysand?

aestene, 2022-10-14:
Note that this only occurs using the remote SQL database and not when using the local SQLite database.

aestene, 2022-10-14:
Note that scheduling a single mission also leaves the robot as busy after the mission has completed so the problem is not dependent on scheduling multiple missions.

oysand, 2022-10-24:
Works now

closed 2022-10-24 by oysand