equinor/flotilla #414: Refreshing mission page renders blank (when deployed in Aurora)

tsundvoll, 2022-10-17:
**Describe the bug**
When clicking on the mission component, redirecting to the mission page works nice. When refreshing the page, the page is rendered blank.

**To Reproduce**
Steps to reproduce the behavior:
0. Test with the Aurora deployment
1. Schedule a mission
2. Click on the mission component
3. Refresh the page

**Expected behavior**
The page should be rendered as when entering it for the first time.

**Screenshots**
If applicable, add screenshots to help explain your problem.

aestene, 2022-12-17:
This issue has automatically been marked as stale as there has been no activity for 60 days. Remove stale label or comment or this will be closed in 7 days.

oysand, 2023-02-20:
Check if this is fixed with configurable homepage address

GodVenn, 2023-05-05:
This is still not working.
The issue is only in Aurora, likely due to the route base always starting with `robotics-frontend` or such.
The issue is very likely related to [parameterized routing in react](https://reactrouter.com/en/main/hooks/use-params), 
since it occurs both when refreshing and when entering the full url directly for both the `/mission/:id` and the `/robot/:id` pages.

aestene, 2023-07-05:
This issue has automatically been marked as stale as there has been no activity for 60 days.

andchiind, 2023-10-12:
Likely relevant: https://github.com/equinor/flotilla/issues/1023