equinor/flotilla #426: Set robot to offline when there is no connextion to ISAR

aestene, 2022-10-19:
**Describe the improvement you would like to see**
Currently the robot status is only updated as there are mission updates from ISAR. However, there is no update when there is no connection to ISAR or ISAR does not have a connection to the robot itself.

ISAR should have a keep alive functionality towards the robot to ensure that the robot is available. This should be included as a keep alive MQTT regular update to Flotilla. Flotilla should then update the robot status based on the keep alive signal. If there is no new MQTT message from ISAR for x time the robot should be set as offline. 

**How will this change existing functionality?**
A clearer understanding of which robots are available and not, for example if a robot has been turned off for the day etc.

**How will this improvement affect the current Threat Model?**
NA

aestene, 2023-01-09:
Will be resolved through https://github.com/equinor/isar/issues/341

GodVenn, 2023-03-27:
Closed by https://github.com/equinor/isar/issues/341

closed 2023-03-27 by GodVenn