equinor/flotilla #433: Error: Failed to GET /echo- missions?installationCode=:

tsundvoll, 2022-10-20:
**Describe the bug**
Missing installation code on get mission.

**To Reproduce**
Steps to reproduce the behavior:
1. At startup of Flotilla. Wait some time and the

**Expected behavior**
200: Returning echo missions

**Screenshots**
[image]

GodVenn, 2022-10-20:
The endpoint should still return 200 even if the code is missing, not sure that is the issue

tsundvoll, 2022-10-20:
Then the issue is more likely related to the previous error: `ERR_NETWORK_CHANGED`

aestene, 2022-12-20:
This issue has automatically been marked as stale as there has been no activity for 60 days. Remove stale label or comment or this will be closed in 7 days.

Christdej, 2023-05-08:
No longer an issue

closed 2023-05-08 by Christdej