equinor/flotilla #441: Use Fusion context for plant-based access control in frontend

GodVenn, 2022-10-25:
**Describe the new feature you would like to see**
Look into using https://www.npmjs.com/package/@equinor/fusion in the frontend for access control and filtering based on plants.

**Describe the solution you'd like**
Fusion's access control system implemented in our frontend

**How will this feature affect the current Threat Model?**

eivindsjovold, 2022-11-10:
Access control in backend, look into using fusion core services https://github.com/equinor/fusion-core-services

tsundvoll, 2022-12-05:
Blocked by Fusion (this feature is under development)

aestene, 2023-06-22:
This issue has automatically been marked as stale as there has been no activity for 60 days.

oysand, 2023-09-18:
Any update on this @eivindsjovold ?

eivindsjovold, 2023-09-18:
I would assume this is mature enough to revisit @oysand .

tsundvoll, 2023-11-06:
Closing for now. Implementing our own access control for now.

closed 2023-11-06 by tsundvoll