equinor/flotilla #456: Present Map in Frontend

tsundvoll, 2022-12-09:
**Describe the new feature you would like to see**
The map should be presented to the user when entering the robot view page / mission page. The map should show the area the robot is going to perform the mission in and a marker displaying where the robot is.

MortFred, 2023-01-06:
Arcitecture discussion for this task 06.01.23 led to initial architecture looking like this:

[image]

MortFred, 2023-01-06:
- [ ] Make Azure blob storage solution with index tags and metadata
- [ ] Implement logic to add a map to new EchoMissions
- [ ] Implement GET RobotMapPosition endpoint
- [ ] Implement GET RobotTagPosition endpoint

PR #556 (2023-02-13): Present map in frontend
Closes #456
closed 2023-02-13 by MortFred