equinor/flotilla #458: Fetch concrete information from backend in the components when needed

tsundvoll, 2022-12-09:
**Describe the improvement you would like to see**

Let each component ask directly for what they need through an apiCaller and dedicated endpoint, instead of asking for for instance a mission on top level and then pass it down into the components. For example for displaying the robot position in `MapPositionSection.tsx` and also for displaying the task table in `TaskTable.tsx`.
A clear and concise description of the improvement proposal.

MortFred, 2023-01-16:
Consider this after finishing implementing #456

phhuseboe, 2023-02-27:
Not blocked anymore @MortFred ?

aestene, 2023-03-13:
@MortFred

anetteu, 2023-04-24:
Closing this issue. A new issue should be created if this is a need in the future.

closed 2023-04-24 by anetteu