equinor/flotilla #484: Estimated mission duration is off

anetteu, 2023-01-04:
**Describe the bug**
The estimated duration of missions is clearly not always correct, as seen from the screenshot. I have not checked whether this is caused by a wrong estimation or invalid start/end times.  

**To Reproduce**
Start flotilla (on commit cd9a9debf7815c1d648ef3f9d12b384530e2be74 when writing this issue), and have a look at the estimated duration of the missions on the front page. 

**Expected behavior**
An reasonable estimated duration of hours. 

**Screenshots**
[image]

oysand, 2023-01-09:
Change estimates from using end time, it should be estimated duration and calculated in the backend

GodVenn, 2023-02-06:
Should be implemented after #554 .
Then we need no math in the frontend, but rather to use the "Estimated Duration" field from the mission object.

Christdej, 2023-02-13:
#554 is implemented

closed 2023-02-22 by Eddasol