equinor/flotilla #485: Naming in front end is not consequent

anetteu, 2023-01-04:
**Describe the improvement you would like to see**
The naming of front end components as `View`, `Section`, `Display`, and similar should be consequent 

**How will this change existing functionality?**
A cleaner and more readable code base

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

oysand, 2023-01-09:
Change function to () => for components as well

aestene, 2023-06-20:
This issue has automatically been marked as stale as there has been no activity for 60 days.

andchiind, 2023-09-20:
This is arguably more of a strategy than a single issue, so it is hard to judge if a PR addresses it.

closed 2023-09-20 by andchiind