equinor/flotilla #498: Move predefined poses from json file to Echo database

oysand, 2023-01-10:
**Describe the improvement you would like to see**
The poses with where the robot shall place itself is currently defined in a local json file and should be uploaded to https://echohubapi.equinor.com/index.html

**How will this change existing functionality?**
New TAGs can now be supported without changing the source code.

**How will this improvement affect the current Threat Model?**
Poses placed in the echo database, may adjust where the robot can move. Who can input these poses should be handled by access control on the echohubapi.

aestene, 2023-06-20:
This issue has automatically been marked as stale as there has been no activity for 60 days.

anetteu, 2023-09-24:
Why is this issue archived from the board and not closed? Should be either closed or open and on the board. Restoring issue and moving to new to force an evalution during planning. I think that this issue can be closed?

oysand, 2023-09-25:
Should be closed

closed 2023-09-25 by oysand