equinor/flotilla #505: Create landing page for non-authorized users

eivindsjovold, 2023-01-13:
**Describe the bug**
Currently, all Equinor employees are authenticated when the log in to frontend. Unauthorized users should be redirected to a landing page where they are asked to apply for access.

**To Reproduce**
Log in to flotilla without being registered as a user in the relevant flotilla Enterprise application(service principal=
**Expected behavior**
Unauthorized users are expected to be redirected to a separate page, and they should not see anything if they naviage to other pages.

**Screenshots**
If applicable, add screenshots to help explain your problem.

eivindsjovold, 2023-01-13:
Related to #504

phhuseboe, 2023-01-13:
Same as #504

PR #618 (2023-03-16): Add Error Boundary and Create Authorization Landing Page
Closes #504 
Closes #505
closed 2023-03-16 by phhuseboe