equinor/flotilla #527: Not possible to cancel/pause ongoing mission in flotilla

anetteu, 2023-01-27:
**Describe the bug**
The cancel and pause buttons for ongoing missions in flotilla did not show. For information, the tasks did not update during the missions either, so there was several things that was not working as expected. 

**To Reproduce**
Ran the newest version form main 26.01.2023 locally on the operator laptop with willliam. 

**Expected behavior**
The pause and stop button should be visible and functional both on the ongoing mission card on the front page and at the ongoing mission page. 

**Screenshots**

[image]

oysand, 2023-02-01:
I was not able to reproduce this. Tried with the latest version and the version of the 26.01.2023 of flotilla, isar and isar-robot. @phhuseboe @eivindsjovold Do you have any idea what this could be?

oysand, 2023-02-02:
Worked after pulling the latest version of ISAR

closed 2023-02-02 by oysand