equinor/flotilla #528: More consistent use of icons

anetteu, 2023-01-27:
**Describe the improvement you would like to see**
I suggest a more consistent use of icons in flotilla, and an alignment with the use of icons in mission planner. Examples are the cancel button in flotilla which looks like a checkbox, and the various delete icons in flotilla/mission planner.

**How will this change existing functionality?**
No change, but hopefully a more intuitive for the user. 

**How will this improvement affect the current Threat Model?**
No change.

Eddasol, 2023-04-22:
I think the checkbox should be changed as we dont have checkbox functionality (possibly after #766 when it can be replaced by order number) but i don’t see ant other inconsistencies. @anetteu  do you agree?

anetteu, 2023-05-31:
This has improved since this issue was created. Closing issue.

closed 2023-05-31 by anetteu