equinor/flotilla #529: Use of capitlal letters should be consistent

anetteu, 2023-01-27:
**Describe the improvement you would like to see**
Use of capital letters should be consistent (only on the first letter of the first word)? Now, there is a mix of for example "Schedule Mission" and "Schedule mission".
[image]

**How will this change existing functionality?**
No change

**How will this improvement affect the current Threat Model?**
No change

aditya22062, 2023-01-27:
# Can be done by multiple methods:
## using split 
## using predefine methods

JoePorterDev, 2023-01-27:
I can take a look at this if it's still available?

Eddasol, 2023-04-22:
I believe this is already fixed. There Are capital first letters in all words for section headlines and table headers and only capital letter in first word for everything else. @anetteu, do you agree that we should close this?

aestene, 2023-06-22:
This issue has automatically been marked as stale as there has been no activity for 60 days.

Eddasol, 2023-06-26:
Already solved

closed 2023-06-26 by Eddasol