equinor/flotilla #530: Task should be arranged from the first to the last task of the mission

anetteu, 2023-01-27:
**Describe the improvement you would like to see**
At the moment, the tasks in the tasks list of an ongoing mission are in random order. Should be arranged from the first to the last, with task number 1 as the first task and following. See screenshot of the task list from the ongoing mission page, where task number 3 is the first task of the mission. No clear identication wether task 1 or task 2 is the next task of the mission. 

[image]

**How will this change existing functionality?**
No change in functionality, but easier to follow ongoing missions in flotilla. 

**How will this improvement affect the current Threat Model?**
No change.

aditya22062, 2023-01-27:
## can be sorted using in progress as key
## can be given serial number which would be easier to understand

tsundvoll, 2023-02-13:
Dependent on issue #567

oysand, 2023-02-20:
We get a sorting order from the echo mission planner

PR #664 (2023-03-21): Common Task and Inspection
Closes #661 
Closes #638 
Closes #635 
Closes #356 
Closes #530 
Closes #406