equinor/flotilla #558: Add overpressure status and warning for relevant robots

anetteu, 2023-02-07:
**Describe the new feature you would like to see**
The overpressure status should be visible for the robots were relevant, possibly with an indication of whether the overpreasure is reaching a limit (check limits defined for relevant robots). The status should also pherhaps be visible in the relevant mission, and a warning pop up should be considered if a robot is approaching a limit/risk of shut down during a mission. 

**Describe the solution you'd like**

**How will this feature affect the current Threat Model?**
No change.

Christdej, 2023-02-13:
Preferably a graph (both for battery and pressure).

anetteu, 2023-03-19:
Close this issue? Should be covered by #673 #674 #675

closed 2023-03-19 by aestene