equinor/flotilla #628: Sorting between maps on various floors

oysand, 2023-02-26:
**Describe the improvement you would like to see**
The map that is selected should also consider the height of the TAGs

**How will this change existing functionality?**
This will support maps that are directly above each other

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

aestene, 2023-06-20:
This issue has automatically been marked as stale as there has been no activity for 60 days.

anetteu, 2023-09-24:
Why is this issue archived but not closed? Restoring and moving to the new  column to force an evaluation during planning. Should either be closed, or open and on the board. Also, this issue should have a better description if it is not closed.

oysand, 2023-09-25:
Should be closed

closed 2023-09-25 by oysand