equinor/flotilla #636: Control buttons inside ongoing mission display 'pause' when it is 'paused', not 'play' as it should

phhuseboe, 2023-03-01:
**Describe the bug**
Control buttons inside ongoing mission display 'pause' when it is paused, not 'play' as it should.

**To Reproduce**
Steps to reproduce the behavior:
1. Start a mission
2. Pause mission from the Ongoing Mission front page section
3. Quickly navigate into the ongoing mission card

**Expected behavior**
Control buttons should always render to 'stop'/'play' when 'paused', and 'stop'/'pause' when 'ongoing'

**Screenshots**
[image]

phhuseboe, 2023-03-01:
Note that this is an edge case. Manually refreshing page solves it, but should be automatic

phhuseboe, 2023-03-10:
Waiting on #505

Eddasol, 2023-03-20:
Pull #662 solves this issue such that the button get updated, but the buttons are still not disabled as the play/pause-switch is ongoing. This makes it possible to choose pause while there already is an ongoing process to pause, leading to an attempt at pausing while the robot is potentially already paused, causing an 502 error.

Clicking pause within the missioncard followed by going to frontpage and stress testing the pause/play-button will cause error

closed 2023-03-20 by Eddasol