equinor/flotilla #660: Order mission queue

Eddasol, 2023-03-09:
**Describe the new feature you would like to see**
Currently the missions get picked from the mission queue in a seemingly random order. It would be better if the mission that got added to the queue first get executed first. 

Bonus: Making sure that the order is presented correctly in frontend as well and allowing the user to modify the order of the missions trough drag-and-drop 

**Describe the solution you'd like**

**How will this feature affect the current Threat Model?**

aestene, 2023-03-13:
Related to #474

aestene, 2023-03-20:
This is how the order is currently displayed

[image]

PR #720 (2023-03-31): Order mission queue by time in backend
Closes #660
closed 2023-03-31 by GodVenn