equinor/flotilla #669: Add MQTT broker mock to test backend MQTT client

GodVenn, 2023-03-16:
**Describe the new feature you would like to see**
Add a mock of the MQTT broker in the backend to allow for testing of MQTT Event Handler behavior.

**Describe the solution you'd like**
A mock MQTT broker publishing topics with mock messages so that we can test different behaviors.
E.g. missing fields in topic message or similar.

**How will this feature affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

mrica-equinor, 2023-07-04:
I need help on this when people are available

aestene, 2023-09-05:
This issue has automatically been marked as stale as there has been no activity for 60 days.

oysand, 2023-10-02:
Try to address the need here with integrations testing instead

closed 2023-10-02 by oysand