equinor/flotilla #699: Scheduled Missions should show immediately in Mission Queue

Christdej, 2023-03-22:
**Describe the bug**
When adding a mission through the "Add Mission" functionallity (see picture below), the mission should be immediately visible in the Mission Queue. Probably because the page is only updated every x seconds, so we should also do this update when pressing Add Mission.
[image]

**To Reproduce**
Schedule a mission through "Add Mission", if you are unluck it can take several seconds before the mission shows in the Mission Queue.

**Expected behavior**
Mission should immediately be visible in the mission queue

GodVenn, 2023-04-13:
We could add it immediately with a loading symbol next to it saying "scheduling..." until the backend confirms it is scheduled maybe?

Christdej, 2023-04-13:
I guess that is good enough, just so that something happends. When nothing happends for several seconds it seems like something went wrong and you try to add another mission before the first is popping up

PR #915 (2023-09-20): Add mockmission for instant feedback in mission queue
closes #699
closed 2023-09-20 by OleDrange