equinor/flotilla #712: Frontend map not great when mulitple tags are stacked

MortFred, 2023-03-26:
**Describe the improvement you would like to see**
It should be possible for the user to see which tags are stacked on top of each other.

[image]

Eddasol, 2023-05-04:
Partially solved by #781. Would also be nice if hovering over a task in the task table moves the tag to the top

aestene, 2023-06-19:
See the Figma sketches for implementation suggestions

Eddasol, 2023-06-19:
[image]

mariannewiggen, 2023-07-03:
Echo has made something similar to this, so suggest doing it like this instead

[image]

Link in figma: https://www.figma.com/file/q6vsNRogzV6r9vTCCDsocd/Robot-Missions?type=design&node-id=4052%3A510657&mode=design&t=0nFqjBsCVHwJMUow-1

aestene, 2023-09-02:
This issue has automatically been marked as stale as there has been no activity for 60 days.

andchiind, 2025-02-13:
This could be partially solved if we implemented a zoom functionality.