equinor/flotilla #722: Show battery/ pressure warning in frontend

UsamaEquinorAFK, 2023-03-31:
**Describe the new feature you would like to see**
Show warning based on the current threshold values set for the robot model. 

**Describe the solution you'd like**
NA

**How will this feature affect the current Threat Model?**
NA

UsamaEquinorAFK, 2023-03-31:
Its blocked by #721

aestene, 2023-07-25:
This issue has automatically been marked as stale as there has been no activity for 60 days.

anetteu, 2023-09-22:
Adding this back to robotics as a new issue to trigger a  prioritization during planning. Why is this no longer on the board?

UsamaEquinorAFK, 2023-09-22:
We should discuss this with ux as well @mariannewiggen

oysand, 2023-09-25:
This is an extension of #893

aestene, 2023-12-18:
Closing as this is covered more extensively by #893

closed 2023-12-18 by aestene