equinor/flotilla #726: Mission tasks random order

eivindsjovold, 2023-04-02:
**Describe the bug**
All tasks in mission no 1

**To Reproduce**
Run mission

**Expected behavior**
Mission identical each time

**Screenshots**
If applicable, add screenshots to help explain your problem.

GodVenn, 2023-04-11:
This is a bug in Echo, they said they had fixed it, but I will contact them again @eivindsjovold

GodVenn, 2023-04-11:
Awaiting fix from echo

aestene, 2023-06-19:
@Eddasol will test if this has been fixed with new missions

Eddasol, 2023-06-19:
This there

Eddasol, 2023-06-19:
Changing the order of some of the tags when creating the mission in echo will prevent this issue

andchiind, 2023-07-24:
This is partly solved in this PR https://github.com/equinor/flotilla/pull/817

oysand, 2023-09-18:
Not experienced in a while, will close the issue and open a new one if needed

closed 2023-09-18 by oysand