equinor/flotilla #730: Quaternions in flotilla backend should be calculated with clockwise rotation, not counterclockwise

phhuseboe, 2023-04-06:
**Describe the bug**
The convertion from Echo's Axis-angle representation to quaternion makes the calculated pose rotate counterclockwise instead of clockwise, making the robot move to wrong pose

**To Reproduce**
Steps to reproduce the behavior:
1. Run mission with taurob or turtlebot
2. Observe that it rotates to wrong pose by doing counterclockwise rotation, not clockwise

**Expected behavior**
The resulting pose should be set following clockwise rotation

**Proposed solution**
Negate the angle used in convertion from Axis-Angle to quaternions

**Screenshots**
N/A

aestene, 2023-04-17:
Uncertain if this issue is related to quaternions and their rotations. Would like a discussion @phhuseboe.

Until then the issue will be kept in TODO.

aestene, 2023-04-17:
Blindzone?

phhuseboe, 2023-06-02:
@aestene It was observed that the base oriented itself north when we told it to orient itself south. We have converted the axis-angle representation of orientation to quaternions, following the right hand rule. Echo's coordinate frame follows the left-hand rule, and I believe that this makes it turn the opposite direction

aestene, 2023-06-09:
We will test this with the robot today.

PR #828 (2023-06-09): Calculate pose with quaternion clockwise rotation
Closes #730
closed 2023-06-09 by mrica-equinor