equinor/flotilla #757: Display failure reason for failed steps

Eddasol, 2023-04-19:
**Describe the new feature you would like to see**
Make failure reason for steps available in flotilla. Possibly as a popover when hovering the failure icon within a mission. This would be very useful for debugging and for understanding the robots behavior.

**Describe the solution you'd like**

**How will this feature affect the current Threat Model?**

aestene, 2023-06-19:
https://github.com/equinor/isar/pull/457 has now been implemented and this issue may be addressed in the frontend.

mariannewiggen, 2023-08-01:
Suggest putting the error messages in the status column in the overview of the missions and adding a "Read more" button if the error message is longer than three lines.

Link to figma sketches: https://www.figma.com/file/q6vsNRogzV6r9vTCCDsocd/Robot-Missions?type=design&node-id=4635%3A618997&mode=design&t=kfBOLaxRlmNfzIr3-1

[image]
[image]

mariannewiggen, 2023-08-02:
It is also possible to update the map with a red color corresponding to statuses that have failed. But perhaps it will be a bit much and that it will be associated with danger, which is in most cases not true

Link to figma: https://www.figma.com/file/q6vsNRogzV6r9vTCCDsocd/Robot-Missions?type=design&node-id=4637%3A632652&mode=design&t=b4SOdQXbJVojHvX7-1

[image]

aestene, 2023-10-01:
This issue has automatically been marked as stale as there has been no activity for 60 days.

andchiind, 2025-05-08:
Closing this as we have moved away from using steps to split up tasks

closed 2025-05-08 by andchiind