equinor/flotilla #766: Order mission queue

Eddasol, 2023-04-22:
**Describe the new feature you would like to see**
The missions in the mission queue should be executed in a predictable order. 

Initially by the order they are scheduled in but with the option to change the order (e.g. by drag and drop i frontend)

**Describe the solution you'd like**

**How will this feature affect the current Threat Model?**

Eddasol, 2023-04-22:
Design inspo for the following frontend task

[image]

aestene, 2023-09-05:
As the focus is moving away from users interacting with individual missions this functionality is not prioritized or necessary and the issue will be closed for now.

closed 2023-09-05 by aestene