equinor/flotilla #769: Add return to home / safe zone functionality

aestene, 2023-04-24:
**Describe the new feature you would like to see**
It should be possible to command the robot to return to a safe zone through Flotilla in cases where the mission should be aborted and the robot should move to a location which is considered secure. A specific example is in case of an alarm where the robot should move out of emergency escape routes it may be blocking. 

**Describe the solution you'd like**
An asset deck should have defined locations which are safe zones that the robot may be commanded to return to. An example of a safe zone could be the docking station where the robot usually lives. 

The frontend should then have a button for the robot which is active when relevant. This button will cancel the currently executing mission and schedule a mission for the robot to return to a defined safe zone. 

**How will this feature affect the current Threat Model?**
NA

Christdej, 2023-05-22:
Needs more work

mariannewiggen, 2023-09-15:
Suggest adding:
- A “Send robot to safe zone” button
- Pop-up for make sure to send robot to safe zone  
- Orange banner for when the robot is on its way to the safe zone
- Green banner for when robot is in safe zone
- Change “Send to safe zone” buttton to “ Safe to continue missions” when robot is on the way to safe sone or in safe zone
- Banner to confirm that the robot has continued with its missions

Link to figma: https://www.figma.com/file/q6vsNRogzV6r9vTCCDsocd/Robot-Missions?type=design&node-id=5318%3A164685&mode=dev

[image]
[image]
[image]
[image]
[image]
[image]

aestene, 2023-11-09:
The final part of this will be handled by #950 so we close this one.

closed 2023-11-09 by aestene